txubelaxu/esphome-jk-bms #47: Slave only reporting at ESP32 restart

davidgp1701, 2025-03-03:
Hello,

I had this running with just 1 JK PB HW version 15 without problem. This JK BMS is configured as master and connected via CAN to a Victron Cerbo GX.

Yesterday I added a second battery, it is connected as a slave in address 0x01, and the ESP32 is connected to its second RS485 port. I can still get all values from the master battery, but the slave only sends data when the ESP32 restarts.

The only thing strange that I can find in the logs is this:

```
[11:58:27][D][jk_rs485_bms:594]: This BMS address is: 1  and address received 0 ==> IDLE
[11:58:27][W][component:237]: Component jk_rs485_sniffer took a long time for an operation (884 ms).
[11:58:27][W][component:238]: Components should block for at most 30 ms.
[11:58:27][D][jk_rs485_sniffer:522]: ..........................................
```

I have attached the full log, in case it is of help.

Both JK BMS are in version 15.38. My configuration is the following one:

```yaml
substitutions:
  esp_name: jk-monitor
  bms0_name: gobel01
  bms1_name: yixiang01
  device_description: "Monitor my JK-BMS via RS485 internal"
  external_components_source: github://txubelaxu/esphome-jk-bms@main #github://syssi/esphome-jk-bms@main
  tx_pin_uart_0: GPIO17 #GPIO1 #GPIO16
  rx_pin_uart_0: GPIO16 #GPIO3 #GPIO17
  talk_pin_rs485: GPIO4

esphome:
  name: ${esp_name}
  comment: ${device_description}
  project:
    name: "txubelaxu.esphome-jk-bms"
    version: "main"

#    version: 2.0.1
#  libraries:
#    - "ArduinoJson"
#  includes:
    #- ./components/ArduinoJson/ArduinoJson.h
    #- ./components/ArduinoJson/ArduinoJson.hpp
    #- ./components/ArduinoJson/Configuration.hpp
#    - ./components/json/json_util.h

esp32:
  board: esp32dev #nodemcu-32s #wemos_d1_mini32
  framework:
    type: esp-idf

external_components:
# - source: ${external_components_source}
#   refresh: 0s
  #- source:
  #    type: git
  #    url: https://github.com/esphome/esphome
  #    ref: dev
  #  components: [ json ]

  - source: ${external_components_source}
    refresh: 0s
    components: [ jk_rs485_sniffer , jk_rs485_bms ]

mdns:
  disabled: true

ota:
  - platform: esphome
    password: "XXXXXXXX"
[... 814 lines cut ...]
    discharging_overtemperature_protection_recovery:
      name: "${bms1_name} discharging overtemperature protection recovery"        
    charging_lowtemperature_protection:
      name: "${bms1_name} charging lowtemperature protection"
    charging_lowtemperature_protection_recovery:
      name: "${bms1_name} charging lowtemperature protection recovery"
    mos_overtemperature_protection:
      name: "${bms1_name} mos overtemperature protection"
    mos_overtemperature_protection_recovery:
      name: "${bms1_name} mos overtemperature protection recovery"      
    cell_count_settings:
      name: "${bms1_name} cell count settings"
    battery_capacity_total_settings:
      name: "${bms1_name} battery capacity total settings"      
    precharging_time_from_discharge:
      name: "${bms1_name} precharging time from discharge"  
    cell_request_charge_voltage_time:
      name: "${bms1_name} cell request charge voltage time"       
    cell_request_float_voltage_time:
      name: "${bms1_name} cell request float voltage time"       
```

Probably an error on my side but I can not figure it out. I have also tried versions `2.0.1` and `2.2`.

[logs_jkmonitor_logs.txt](https://github.com/user-attachments/files/19052367/logs_jkmonitor_logs.txt)

txubelaxu, 2026-08-18:
Thanks for the detailed report and the log attachment. Two bugs were just fixed on `main` (commit `58cf2c0`) that look like a good match for this:

- The RX buffer had no real growth cap - it was using `vector::max_size()` as the limit, which is effectively unbounded (the theoretical allocator limit, not a practical one). A buffer that keeps growing without ever being trimmed would explain exactly the kind of `Component jk_rs485_sniffer took a long time for an operation (884 ms)` warning in your log, since parsing/searching cost grows with buffer size.
- `publish_state_(sensor::Sensor*, float)` had a guard that silently dropped any `NaN` value, so a slave's sensors never correctly went "unavailable" when it stopped responding - they'd just freeze at their last value instead of recovering. That matches "slave only sends data when the ESP32 restarts" quite well.

Both are on `main` now. Would you be able to test against current `main` and see if the second slave now keeps reporting without needing a restart?

txubelaxu, 2026-08-19:
@/tmp/comment_47_followup.md