txubelaxu/esphome-jk-bms #48: JKPB2A16S One master 2 slave last version issues

Mansionus, 2025-04-19:
Hello, thanks a lot for the great job, i have tried several version and on this configuration based on the esp32-example-jkpb-rs485_1master_6slaves.yaml adapted for 1Master and 2 Slaves and removing cell 17 to 24. 

I have also tried several settings and from version 2.0.1 this is not working well the slave are updated at the start points and after they are either unknow after some time or updated ramdomly but not both slave.

From version v1.3.1 this is working much better and all BMS are refreshed frequently.

What am I doing wrong with the new version ?

Thanks

rabbit3dcustom, 2025-07-12:
> [@rabbit3dcustom](https://github.com/rabbit3dcustom) i see that you forked this repo and made some changes. Did you manage to get this working? I self got 1 master and 1 slave. Master is no problem but I cant see the slave at all.
> 
> I tried different things for 3 evenings already but can't find the issue.

I'm trying.. slowly, till I get used to the way of the coding and all the stuff.

Right now i'm able to maintain the ESP32 connected and working almost 2 days long with two cable-connected JK-PB (no master, both slaves).

Thanks a lot for your interest! so appreciate.

rabbit3dcustom, 2025-07-23:
> [@rabbit3dcustom](https://github.com/rabbit3dcustom) i see that you forked this repo and made some changes. Did you manage to get this working? I self got 1 master and 1 slave. Master is no problem [...]

@LithiumNL  Could you please go to my github and in the issue section, post your issue? so i can check it.

txubelaxu, 2026-08-18:
Thanks for the report, and for narrowing it down to a regression somewhere between v1.3.1 and v2.0.1+ - that's a useful data point. Two bugs were just fixed on `main` (commit `58cf2c0`) that match your symptoms well:

- The RX buffer had no real growth cap (it used `vector::max_size()`, effectively unbounded instead of a practical limit).
- `publish_state_(sensor::Sensor*, float)` had a guard that silently dropped any `NaN` value, so a slave's sensors never correctly went "unavailable" when it stopped responding - they'd just get stuck at their last value instead of cleanly recovering. That fits "one or both slaves become unknown" and "sometimes one slave gets updates, but never both consistently" quite well.

Both fixes are on `main` now. Could you try current `main` again with your 1 master + 2 slaves setup and see if it stays consistent this time?