txubelaxu/esphome-jk-bms #49: Single battery, mode 1 PSEUDO-MASTER

copilador, 2025-04-23:
Hi,

I would like to use Mode 1 with only one JK-BMS (a single battery setup), but I’m not sure which configuration or example code would be appropriate for that scenario.

Should I set the BMS address to 0x00 in order to enable Mode 1 (sniffer + pseudo-master mode)?

I want to monitor this one BMS without the ESP32 acting as the master.

Any guidance or example would be greatly appreciated.
Thanks in advance!

txubelaxu, 2026-08-19:
@/tmp/comment_49.md

txubelaxu, 2026-08-20:
Yes — Mode 1 is exactly what you want, and a single BMS is fine (there's no minimum device count).

1. Set the physical JK-BMS's DIP switches to address `0x00` so it's the REAL MASTER on the RS485 bus.
2. In your YAML, configure a single `jk_rs485_bms:` entry with `rs485_address: 0x00`, pointing at your `jk_rs485_sniffer`. No slave entries needed.

You can base it directly on [`esp32-example-jkpb-rs485_1master_1slave.yaml`](https://github.com/txubelaxu/esphome-jk-bms/blob/main/esp32-example-jkpb-rs485_1master_1slave.yaml) — just delete the second (`bms1`, address `0x01`) block and keep only the `0x00` one.

With this setup the ESP will passively sniff the master's own broadcast traffic (cell info, settings) and periodically act as a temporary pseudo-master only to fetch the one frame type the master never broadcasts on its own (device info: name, hw/sw version, password, RCV/RFV time) — see the "MODE 1" section in the [README](https://github.com/txubelaxu/esphome-jk-bms/blob/main/components/jk_rs485_bms/README.md#mode-1-if-there-is-a-jk-bms-acting-as-real-master). The ESP itself never becomes the bus master in this mode, which matches what you're after.